pimoroni/ioe-python #2: Rotary Encoder example didn't work with DFRobot EC11

brian-burton, 2020-08-29:
Hi,

I tried the Rotary encoder example with a DFRobot EC11, and didn't receive any interrupts from the IO Expander until I commented out [this line](https://github.com/pimoroni/ioe-python/blob/ea84232b6dbaedefa12a832d09e8ac90e0e4741e/examples/rotary.py#L26), then everything worked fine.

Gadgetoid, 2020-09-10:
Thanks- I'll have to add some documentation to clarify this.

The rotary example is actually intended to run a breakout we've got in the pipeline that has a rotary encoder onboard.

Good to know you managed to get it up and running, though!

KeeCie, 2020-10-02:
Hi,

I think this is not the real solution. I think the following is missing in the example:
ioe.set_pin_interrupt(pin, enabled=True)         (with pin, the pin you use on the IO Expander)

Now the interrupt worked for  #me. I tried it with the button example (see code below):

import time
import ioexpander as io

print("""button.py

Demonstrates handling a single button input using IO Expander.

Your button should be wired between pin 14 and ground on the IO Expander.

A pull-up resistor will be enabled, causing your button to read 1 normally, and 0 when pressed.

Press Ctrl+C to exit.

""")

BTN_PIN = 14

ioe = io.IOE(i2c_addr=0x18)
ioe.set_mode(BTN_PIN, io.IN_PU)

ioe.enable_interrupt_out()
ioe.set_pin_interrupt(14, enabled=True)
ioe.clear_interrupt()

last_value = io.HIGH
last_inter_value = io.LOW

while True:
    value = ioe.input(BTN_PIN)
    inter = ioe.get_interrupt()
    if inter != last_inter_value:
        print("Interrupt:",inter)
        last_inter_value = inter

    if value != last_value:
        print("Button has been {}".format("released" if value else "pressed"))
        last_value = value
        ioe.clear_interrupt()

    time.sleep(1.0 / 30)

Succes experimenting.

brian-burton, 2021-05-26:
Looks like this was fixed in #3 , thank you! #

closed 2021-05-26 by brian-burton